ngageoint/sarpy #1: Please make pip package and/or more elaborate ReadMe for usage

kathomas921, 2018-03-29:
Could you create a python package or at least a wheel for this so people can pip install? The ReadMe also gives very little guidance at the moment as to how to actually use the code.

Without a package, I tried to navigate into the io/complex and follow the "Example of how to use this package" steps, but I'm getting an error saying the following because I am trying to import from geometry, which is above my current root folder of io.

`>>> import complex
>>> reader_object = complex.open('test.h5')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "complex/__init__.py", line 70, in open
    modules = [sys.modules[names] for names in module_names if __import__(names)]
  File "complex/radarsat.py", line 4, in <module>
    from .sicd import MetaNode
  File "complex/sicd.py", line 8, in <module>
    from ...geometry import geocoords as gc
ValueError: Attempted relative import beyond toplevel package`

If I go up a level and import io.complex as complex, I'm getting the following error, which I don't understand as well.
`>>> import io.complex as complex
>>> reader_object = complex.open('test.h5')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "io/complex/__init__.py", line 70, in open
    modules = [sys.modules[names] for names in module_names if __import__(names)]
  File "io/complex/radarsat.py", line 4, in <module>
    from .sicd import MetaNode
  File "io/complex/sicd.py", line 6, in <module>
    from .utils import bip
  File "io/complex/utils/bip.py", line 3, in <module>
    import numpy as np
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import add_newdocs
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/lib/__init__.py", line 23, in <module>
    from .npyio import *
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/lib/npyio.py", line 14, in <module>
    from ._datasource import DataSource
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/lib/_datasource.py", line 220, in <module>
    _file_openers = _FileOpeners()
  File "/home/kate/miniconda3/envs/py27/lib/python2.7/site-packages/numpy/lib/_datasource.py", line 162, in __init__
    self._file_openers = {None: io.open}
AttributeError: 'module' object has no attribute 'open'
>>> reader_object = complex.open('test.h5')`

I'm excited to use the package once I get it up and running. Thanks!

utwade, 2018-04-05:
Sorry, I posted the initial code quickly without a lot of time for proper packaging and then ran off to spring break, so I'm late on replying to this.  I agree that some examples are needed.  I will try to get some examples posted soon to clarify this.

ayoungs is correct that it will not currently handle HDF5 files (like CosmoSkymed).  The architecture is intended to handle complex SAR data, primarily in SICD format (as can be found [here](https://github.com/ngageoint/six-library/wiki/Sample-SICDs).)

However, this library also allows for the defining other complex SAR formats/readers (which get converted to SICD internally, so to the calling code they will look just like a SICD with the same metadata structure, data indexing, etc.) Thus far, besides SICD, only Sentinel-1 and RADARSAT-2/RCM native formats are handled.  We could certainly add the CosmoSkymed SCS/SLC format in HDF5.  We have the code in MATLAB to convert it to SICD (It will be posted soon [here](https://github.com/ngageoint/MATLAB_SAR)), so it would be a relatively simple port.

utwade, 2018-04-10:
I've added some more proper Python wrapping to the package to include a setup.py to allow for creation of wheels, source distribution, etc.  Although you can also just simply add the sarpy subdirectory to your Python path, and everything should run as its "universal" in Python terminology (pure Python and runs without any changes on Python 2 and 3.)

Additionally, I've added a sarpy_example.py file in the docs directory for a simple example of how to use the SarPy code to read a file and inspect its metadata and pixels.  Hope this helps.

closed 2018-04-10 by utwade