ngageoint/sarpy #2: AttributeError: 'module' object has no attribute 'BufferedIOBase'

apeternier, 2018-04-19:
I used setup.py build and install to have things properly installed under Ubuntu 16.04.4. I then tried to run the sarpy_example.py but I'm stuck in this error: 

AttributeError: 'module' object has no attribute 'BufferedIOBase'

After some reading, it looks to be a conflict with the 'io' directory within 'sarpy':

https://stackoverflow.com/questions/32657580/class-gzipfileio-bufferediobase-attributeerror-module-object-has-no-attrib?noredirect=1&lq=1

I'm really not familiar with python and I don't know how to fix similar issues. It seems that renaming such 'io' directory to something else should solve the problem, but then all the code and package paths should be updated accordingly.

utwade, 2018-04-19:
As long as you aren't importing 'io' by itself, I don't think this should be an issue, since Python generally keeps namespaces well organized and separated by packages/modules.  You should explicitly have to reference sarpy.io, rather than just io, at least in the way I have traditionally used it.

Also its possible that the issue is with the setup install, which to be honest, I don't have a lot of experience in, and am open for input if its done wrong here.  In general, I have never used that for sarpy.  I just add the sarpy directory to my Python path.  Although I believe the setup install should work as well, so if it doesn't, we should track that down.

Note that sarpy_example.py wasn't really intended to be run as a whole by itself, but just give code snippet illustrations of how to do some basic file parsing.  At the least, you will have to change the "fname = ..." line to point to a SICD/RS2/Sentinel dataset that you have.

Perhaps you could send a fuller stack output of your error to help track down the issue?

Thanks for the feedback.

utwade, 2018-04-19:
Upon further inspection, you are absolutely right.  The setup file is not installing the sarpy package as a whole, but rather the level of packages beneath that.  I will fix it and upload.

In the meantime, you can just add sarpy to your Python path to get around the setup script problem.

utwade, 2018-05-10:
Just uploaded what I believe is a fix to setup.py for the install.  Please let us know if it doesn't work as you expect it. Sorry for the delay.

It should now require you to access sarpy.io, rather than io by itself, which was causing name collisions, resulting in the errors you saw.  The intent was alway to access things through a sarpy heirarchy, rather than standalone subpackages.

closed 2018-05-10 by utwade