ngageoint/sarpy #9: Deprecation Workaround for scipy.misc.comb

elwell67, 2019-09-18:
`scipy.misc.comb` is used in csk.py, radarsat.py, and sentinel.py under io/complex/utils, and maybe others.  This scipy function has been deprecated since version 1.0.0 ([documentation here](https://docs.scipy.org/doc/scipy-1.2.1/reference/generated/scipy.misc.comb.html)).

The following solution (from [here](https://stackoverflow.com/questions/27531361/calculating-the-number-of-k-combinations-with-and-without-scipy)) is an alternative that requires no external dependencies:  

`from math import factorial` 
`def comb(n, k):`
`return factorial(n) / factorial(k) / factorial(n - k)`

utwade, 2019-09-18:
Agreed.  We will fix.

jcasey-beamio, 2019-09-20:
a PR has been submitted to address this issue.  The implemented fix checks the scipy version number and imports from scipy.special if the scipy version is 1 or higher, and from scipy.misc otherwise.  It should work with new versions of scipy and provide backward compatability for older versions.

closed 2019-09-24 by utwade