ngageoint/sarpy #54: No module 'sarpy.io'

khavernathy, 2020-04-16:
With a clean install, the sarpy python modules don't seem to load properly. Please assist

```
$ git clone https://github.com/ngageoint/sarpy/
$ cd sarpy
$ python3 setup.py install --user
$ cd docs/examples
$ python3 sarpy_example.py
Traceback (most recent call last):
  File "sarpy_example.py", line 6, in <module>
    import sarpy.io.complex as cf
ModuleNotFoundError: No module named 'sarpy.io'
```

khavernathy, 2020-04-16:
I can import `sarpy` but not `sarpy.*` ..

```
$ ipython3
Python 3.6.7 | packaged by conda-forge | (default, Nov  6 2019, 16:19:42) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.8.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import sarpy

In [2]: import sarpy.io
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-2-5599829837e9> in <module>
----> 1 import sarpy.io

ModuleNotFoundError: No module named 'sarpy.io'
```

tommccullough-tenica, 2020-04-16:
I have the packages poorly discovered, and will have a fix in short order.

tommccullough-tenica, 2020-04-16:
Douglas, the master branch is repaired, and you should be good to go

closed 2020-04-16 by tommccullough-tenica

khavernathy, 2020-04-17:
all set, thanks!

tommccullough-tenica, 2020-04-17:
If you have any trouble that doesn't really rise to the level of creating an issue, feel free to contact me directly via email at thomas.mcculough at tenica-gs dot com.