ngageoint/sarpy #62: add sarpy to PyPI repository?

khavernathy, 2020-05-15:
Would it be possible to add sarpy to PyPI so that it can more easily be used as a dependency in other python projections, e.g. with version specificity etc.? https://pypi.org/

tommccullough-tenica, 2020-05-18:
This is something that we have discussed, and it will be done is the relatively near future. There is a little delay here because we want to split this repo appropriately. There are really 3 different  packages in this one repo, and this will be split into three separate repositories. Doing this is on the task list, but not yet complete.

In the short term, you can specify the dependency for your project, you just have to manually build the wheel and install for sarpy. This is clearly not the appropriate long term solution, but is easily done.

You can build the wheel using the command `python setpy.py bdist_wheel` and the wheel will be built and placed in the `dist` folder. Then, install the wheel using `pip install <wheel file>`. 

Alternatively, if you use one of the "short cut" install commands, like `python setupy.py install --user`, then the version number should still be recognized from the egg, but I've never tried. You can always tell the version number of the sarpy that you are using via `sarpy.__version__` at the python prompt.

khavernathy, 2020-05-18:
Sounds good, thanks!

tommccullough-tenica, 2020-06-30:
This is complete, and will be maintained for updates moving forward.

tommccullough-tenica, 2020-07-02:
sarpy has been uploaded to PyPI

closed 2020-07-02 by tommccullough-tenica

khavernathy, 2020-07-10:
Great news, thank you