ngageoint/sarpy #92: read_chip unexpected behavior

darling-imsci, 2020-07-28:
When I read a chip using slices, I get what I expect:
```
import sarpy.io.complex as cf

ro = cf.open(fn) #some .sicd file
ro[0:100,0:100].shape
(100, 100)
```
When I use the read_chip function without the 3rd parameter in each bound range defined, I would expect to get the same output as above:
```
ro.read_chip((0,100),(0,100)).shape
(100, 100)
```
Instead, I do not get any data back:
```
ro.read_chip((0,100),(0,100)).shape
(0, 0)
```
If I include the 3rd parameter, things work the way I would expect again
```
ro.read_chip((0,100,None),(0,100,None)).shape
 (100, 100)
```
I see the note under the `_reorder_arguments` function that says that if range = (int,int), then range = (stop, step), and I can get around this by just always making the 3rd argument in the bounds = 1, but this seems like unintuitive behavior.

tommccullough-tenica, 2020-07-28:
Hello, I agree that this is not necessarily intuitive behavior. The standard slicing/fetching call is included for standard Pythonic syntax and convention.

The long story short is that the initial development of sarpy is actually motivated by a long standing MATLAB project, in particular, the read_chip function is directly translated from the conventions in that project. That is the convention expected by this pre-existing community.

Below pulled out of the docstring for the read_chip function:
`dim1range : None|int|Tuple[int, int]|Tuple[int, int, int]
  The row data selection of the form [start, [stop, [stride]]], and
  None defaults to all rows (i.e. (0, NumRows, 1))
`
Please let me know, what could I do to make it more clear that argument (0, 100) is equivalent to stop=0, stride=100, and stop=0, stride=anything will yield no results?

darling-imsci, 2020-07-28:
I guess I've never seen the notation quite like the docstring ( [start, [stop, [stride]]]) where the first item is the least significant, and I didn't know what to think of it.
In that case, I think just adding an extra example at the end of the read_chip docstring of these two partial cases might work:
``` 
data = reader.read_chip((start1, stop1, stride1), (start2, stop2, stride2))
data = reader.read_chip((stop1, stride1), (stop2, stride2))
data = reader.read_chip((stride1), (stride2))
```

tommccullough-tenica, 2020-07-28:
Thanks for your feedback. I will follow your suggestion and include this example in the docstring. This issue will be closed when this is merged in.

closed 2020-07-28 by tommccullough-tenica