ngageoint/sarpy #99: sicd_meta.RadarCollection.Area.Plane.SegmentList Failed converting XML

darling-imsci, 2020-08-05:
I have a SICD with with a RadarCollection.Area.Plane segment defined in the DES, but I get an error when I try loading the file using sarpy. 
I created this example XML (modified from the SIX repo) which validates against the XML schema:
[sicd_example_RMA_RGZERO_RE32F_IM32F_cropped_multiple_image_segments_PlaneArea-XML_DATA_CONTENT0.txt](https://github.com/ngageoint/sarpy/files/5030385/sicd_example_RMA_RGZERO_RE32F_IM32F_cropped_multiple_image_segments_PlaneArea-XML_DATA_CONTENT0.txt)

```
xmllint --schema sarpy/sarpy/io/complex/sicd_schema/SICD_schema_V1_1_0_2014_07_08.xsd sicd_example_RMA_RGZERO_RE32F_IM32F_cropped_multiple_image_segments_PlaneArea-XML_DATA_CONTENT0.txt > xmllint.txt
sicd_example_RMA_RGZERO_RE32F_IM32F_cropped_multiple_image_segments_PlaneArea-XML_DATA_CONTENT0.txt validates
```
When I manually read this XML using sarpy, I get an error:
```
>>> from sarpy.io.complex.sicd_elements.SICD import SICDType
>>> with open('sicd_example_RMA_RGZERO_RE32F_IM32F_cropped_multiple_image_segments_PlaneArea-XML_DATA_CONTENT0.txt','rb') as f:
>>>      des_bytes = f.read()
>>> root_node, xml_ns = utils.parse_xml_from_string(des_bytes.decode('utf-8').strip())
>>> sicd_meta = SICDType.from_node(root_node, xml_ns, ns_key='default')
ERROR:root:Failed converting <Element '{urn:SICD:1.1.0}Plane' at 0x7f04263467c8> of type <class 
xml.etree.ElementTree.Element'> to Serializable type <class 
'sarpy.io.complex.sicd_elements.RadarCollection.ReferencePlaneType'> for field Plane of class AreaType 
with exception <class 'ValueError'> - Attribute coords of array type functionality belonging to class 
SerializableArray got a ElementTree element with size attribute 3, but has 0 child nodes with tag 
SegmentList.. Setting value to None, which may be against the standard.
```
It looks like if you change sarpy.io.complex.sicd_elements.RadarCollection.ReferencePlaneType._collections_tags to {'SegmentList': {'array': True, 'child_tag': 'Segment'}}, it seems to import.

darling-imsci, 2020-08-05:
It also looks like, from the schema, "TypeId" in MatchInfo.MatchType should be "TypeID"

tommccullough-tenica, 2020-08-06:
Thanks for supplying the file in question. I'll rectify this issue and any others that I find in the morning.

tommccullough-tenica, 2020-08-06:
Many thanks for finding the two bugs

closed 2020-08-06 by tommccullough-tenica