ngageoint/sarpy #119: Invalid SICDs when using numpy 1.9

bombaci-vsc, 2020-09-03:
setup.py specifies the [minimum version of numpy to be 1.9](https://github.com/ngageoint/sarpy/blob/master/setup.py#L33), however the SICD converters do not work correctly with 1.9.

I created a RS2 SICD and the resulting `xsd:dateTime` fields are improperly formatted.  They contain my local time zone in addition to the `Z`.  As a result the XML does not pass the schema.
```xml
        <DateTime>2015-11-30T18:13:53.000000-0700Z</DateTime>
```
```xml
        <CollectStart>2008-05-04T01:30:48.413940-0700Z</CollectStart>
```

This looks to be caused by the [datetime behavior change introduced in numpy 1.11](https://numpy.org/doc/stable/release/1.11.0-notes.html#datetime64-changes).

The simplest fix is probably to change the minimum numpy version to 1.11.  If 1.9 is still needed, then the timezone needs to be stripped out, maybe in [`serialize_plain`](https://github.com/ngageoint/sarpy/blob/master/sarpy/io/complex/sicd_elements/base.py#L1710)?

tommccullough-tenica, 2020-09-03:
I agree that the simplest fix is just to bump the required numpy version. That may not be an acceptable solution for the decision makers, who are trying to be reasonably version agnostic in a nod towards difficulties caused by cyber security policies and the like. There will be a discussion about appropriate path forward. 

Regardless of the decision, a fix should be completed by early next week.

tommccullough-tenica, 2020-09-03:
Thanks for bringing this to our attention. The decision was made to bump the numpy dependency version to 1.11.

closed 2020-09-03 by tommccullough-tenica