ngageoint/sarpy #123: problem running fft-example

bradh, 2020-09-14:
```
bradh@minutae:~/sarpy/docs/examples$ python3 ./fft_example.py 
File /home/bradh/Data/sarpy_data/nitf/sicd_example_1_PFA_RE32F_IM32F_HH.nitf is determined to be a SICD (NITF format) file.
ERROR:root:ImpRespBW (15.17029) must be <= DeltaK2 - DeltaK1 (15.17028)
ERROR:root:Issue discovered with Col attribute of type <class 'sarpy.io.complex.sicd_elements.Grid.DirParamType'> of class GridType.
ERROR:root:Issue discovered with Grid attribute of type <class 'sarpy.io.complex.sicd_elements.Grid.GridType'> of class SICDType.
compute the fft to display in range / polar azimuth
Traceback (most recent call last):
  File "./fft_example.py", line 15, in <module>
    cdata = ro.read_chip((0, ro.data_size[0] - 1, 1), (0, ro.data_size[1] - 1, 1))
TypeError: unsupported operand type(s) for -: 'tuple' and 'int'
```

it looks like its opening the file OK, but that `read_chip` is returning an tuple of tuple: `((3975, 6724),)`.

It works better if I make this change:
```
diff --git a/docs/examples/fft_example.py b/docs/examples/fft_example.py
index 09f7c55..d7e79ae 100644
--- a/docs/examples/fft_example.py
+++ b/docs/examples/fft_example.py
@@ -12,7 +12,7 @@ fname = os.path.expanduser(os.path.join('~/Data/sarpy_data/nitf', 'sicd_example_
 ro = cf.open(fname)
 
 print("compute the fft to display in range / polar azimuth")
-cdata = ro.read_chip((0, ro.data_size[0] - 1, 1), (0, ro.data_size[1] - 1, 1))
+cdata = ro.read_chip((0, ro.data_size[0][0] - 1, 1), (0, ro.data_size[0][1] - 1, 1))
 
 cdata = cdata[0:1000, 0:1000]
```

I'm not familiar with the code base, so can't say if I'm fixing or working around the actual issue here. Let me know if a PR would help.

tommccullough-tenica, 2020-09-14:
Thanks for pointing this out. Unfortunately, this is an outdated example. 

This part of the project honestly needs more attention. In the short term, the example scripts will get reviewed and updated over the next few days.

tommccullough-tenica, 2020-09-17:
This example was deprecated, and was deleted. I updated the example usage in the basic example.

closed 2020-09-17 by tommccullough-tenica