ngageoint/sarpy #125: Embedded records in ENGRDA TREs not parsed correctly

ianrcooke, 2020-09-17:
When trying to process a NITF with an ENGRDA TRE, I'm finding that the embedded Records fail to parse.  Rather than starting at an offset into the TRE data (at ENGLN, 23 bytes in, after RESRC and RECNT), they're starting at the beginning (i.e. the RESRC) field and failing because because what's parsed is not an integer, as expected, but character data.

This is possibly a bug in [TRELoop]( https://github.com/ngageoint/sarpy/blob/dac6ac49313c4789175de307de4bff1082c4e54a/sarpy/io/general/nitf_elements/tres/tre_elements.py#L215).  Changing 

```python
 entry = child_type(value, *args, **kwargs)
```

to 

```python
 entry = child_type(value[loc:], *args, **kwargs)
```
fixed this for me locally.

tommccullough-tenica, 2020-09-17:
Thanks for your bug finding. Coincidentally, someone pointed these two and a few others out to me a couple hours ago. I will have a fix for these and others in by tomorrow afternoon.

tommccullough-tenica, 2020-09-18:
This should be fixed in the latest merge

closed 2020-09-18 by tommccullough-tenica