ngageoint/sarpy #154: Update tests with URLs of missing test data

RubeRad, 2020-11-19:
Right now just cloning the repo and 'setup.py test' fails all over the place. The error messages make clear where the test data is expected to be found, but not where a user can download it from in order to put it there.

The link in the README about the sample ntf is helpful, but not complete. I also had to hunt on my own to find required test data at

- https://gwg.nga.mil/ntb/baseline/software/testfile/Nitfv2_1/scen_2_1.html
- https://geographiclib.sourceforge.io/html/geoid.html
- ftp://data.asc-csa.gc.ca/users/OpenData_DonneesOuvertes/pub/RCM/Montreal/RCM1_OK1049807_PK1049808_1_16M11_20191028_224339_HH_HV_SLC/

and I still can't find data for the Sentinel and RS2 tests. 

It would be good if each test failure message would say like "The data needed for this test can be downloaded from http://..."

Even better, how about if there were a 'setup.py download' that automatically downloaded and organized all the required test data? (with a confirmation like 'This will download X gigabytes of test data into ...\Desktop\sarpy_testing, is that OK? [Y/n]' )

tommccullough-tenica, 2020-11-19:
Hi RubeRad, thanks for your participation. Without some internal hosting capability which we do not currently have, and some of the data may potentially have licensing issues, the idea of an automatic download is probably infeasible. 

You are certainly correct that the information needs to be updated, and more helpful messages logged. I will also ensure that tests which cannot be performed do not result in a failure.

RubeRad, 2020-11-19:
Yeah I just meant downloading from whatever external URLs the data hopefully lives at, but a trail of breadcrumbs in the failure messages is a start.

Are the RS2/Sentinel data online that you know of?

tommccullough-tenica, 2020-11-19:
I was able to find sentinel data from alaska/nasa data provider: https://search.asf.alaska.edu/#/?resultsLoaded=false&zoom=3&center=-97.493959,39.672786&view=equitorial&productTypes=SLC

You should be able to get access to some, but you have to request and account or something - it's been awhile since I did it. I've honestly never tried to find open radarsat-2 data. 

To be completely clear, these unit tests are really just for me as a developer for constructing the various reader types and making sure that I don't break anything. You really don't typically need to run these as a user - there are no configuration issues or anything. Obviously, you are more than welcome to do so, however.

tommccullough-tenica, 2020-11-19:
You can get some sample data for radarsat-2 here: https://mdacorporation.com/geospatial/international/satellites/RADARSAT-2/sample-data/

The complex format data is the SLC data.

RubeRad, 2020-11-19:
Awesome, thx for the pointers, I will check those out and see if I am able to use them to get 11/11 on the tests.

If you want, you can drop me a ping to my same username at gmail; I'm happy to discuss my affiliation and business with sarpy offline.

tommccullough-tenica, 2020-12-13:
I updated the structure of the tests, and put a read me in the tests directory. It's now easier to configure, at least. I'm closing this for now, but this remains a work in progress.

closed 2020-12-13 by tommccullough-tenica