ngageoint/sarpy #173: SICDReader to_xml_string() changing SICD element to SICDType

johnhhenderson, 2021-01-20:
I need to extract the SICD XML string from a SICD file.  I will also be using the pixels, so want to use the SICDReader to do it all.  I assume I am using it correctly - I am effectively doing:
sicdXml = SICDReader("testImage.ntf").sicd_meta.to_xml_string().
The resulting string gave me an error in my proprietary code.  Before attempting to use sarpy, I was using a proprietary sicd string reader.  Upon examination, I noticed that the the top-level tag name had been changed from "SICD" to "SICDType" in the string returned by sarpy.  Is this intentional?  It would be convenient if to_xml_string() returned the XML string as it appears in the file.  
Making a cursory glance at the rest of the returned string, I do not see any other element name substitutions.

tommccullough-tenica, 2021-01-20:
Change your command to SICDReader("testImage.ntf").sicd_meta.to_xml_string(tag='SICD').

This is essentially just a foible of my class inheritance, and I should fix it.

johnhhenderson, 2021-01-21:
Thank you, Tom.  That's a little neater solution in the meantime than my string replace.

closed 2021-02-02 by tommccullough-tenica