ngageoint/sarpy #182: example for writing a SICD?

khavernathy, 2021-02-15:
Is there an example for writing a SICD in the source code? Having trouble finding something like

`import sarpy.io.complex.SICDWriter`
`SICDWriter(image=image, meta=meta, filename='out.sicd')`

khavernathy, 2021-02-15:
ah I got something

```
from sarpy.io.complex.sicd import SICDWriter as SICDWriter
x = SICDWriter('out.sicd', meta)
x.write_chip(cimg_small, (0,0), 0)
```

closed 2021-02-15 by khavernathy

tommccullough-tenica, 2021-02-15:
Please feel free to email me questions directly at thomas.mccullough@tenica-gs.com. I should create an example for writing a SICD, to make it simple and clear. Are you open to collaborating on such an example?

In the code base, the only place that a sicd is getting written is in the conversion from one format to another. It's a bit convoluted to simply point at. If you send me an email, I'll flesh out the basics of writing a sicd. What you just sent is essentially it. I would recommend using it as a context manager (with SICDWriter() as writer: ...).