ngageoint/sarpy #207: SICDWriter: add required "SICD:" characters to NITF FTITLE

johnhhenderson, 2021-03-17:
Volume 2 of the SICD 1.1 specification, table 3-2, "NITF 21. File Header" indicates that the FTITLE field will be "SICD: <core name>."  SARWriter does not automatically prepends the "SICD: " but preferably should.
In our use, some applications reading the SICD created with sarpy use the FTITLE field to identify the file type, and fail because the string is not included.
We are currently working around the problem by prepending the CoreName with "SICD: ," but this is, of course, changing the CoreName field, which we do not want to do, as it is often used to identify the collection.

johnhhenderson, 2021-03-17:
Need to investigate issue further. Sorry for the mess.

closed 2021-03-17 by johnhhenderson

tommccullough-tenica, 2021-03-17:
No trouble. Just so you know the origins, I originally had it set to "SICD: <corename>" as suggested in the standards document. Someone gave me some specific guidance to use something else in many cases, and that's what's there now. I need to have a discussion with my POC, but I'm probably open to changing it back.

I feel like this is disappointing, but I've convinced myself that the only *reliable* way of ensuring that the file is actually SICD file is to check:
1.) there are data extensions present - SICD files could have tre overflow data extensions or other additional data extension, but probably only have exactly one data extension. Note that SIDD files are permitted to have SIDD data extension(s) and SICD data extensions.
2.) exactly one of these data extensions contains a SICD structure - again, note that SIDD files may contain SICD data extension(s) in addition to SIDD data extension(s). 
3.) the image data is complex and in keeping with the SICD structure - SICD files will contain complex data, but SIDD files will definitely not. Some other NITF files may contain complex image data, but not the SICD data extension - these will probably have meta data scattered through a variety of tres.

That said, I'm trying to be completely thorough, and there's likely something less complicated that probably works in most cases. Feel free to email any correspondence at thomas dot mccullough dot ctr at nga dot mil.

johnhhenderson, 2021-03-17:
After I wrote the ticket, I looked at the code, and I saw a place where the “SICD:” was being added to the FTITLE (_get_ftitle()), so I wasn’t sure about my observation, so I withdrew it, but it does appears that in the way I’m using SICDWriter, it is not being added.

In general, if there’s a specification, I would like to see compliance with the specification.

We have delivered software with customers that expect SICD specification compliance and check for that “SICD:” string in the FTITLE.  The programs I am supporting now cannot influence changes to those products.  If there is software in use from other vendors that are looking for that, I certainly cannot ask them to change the code.

Thanks,
John

  

CONFIDENTIALITY NOTICE: This email and any attachments are for the sole use of the intended recipient and may contain material that is proprietary, confidential, privileged or otherwise legally protected or restricted under applicable government laws. Any review, disclosure, distributing or other use without expressed permission of the sender is strictly prohibited. If you are not the intended recipient, please contact the sender and delete all copies without reading, printing, or saving.

utwade, 2021-03-17:
Actually, the "SICD: " requirement no longer exist in the latest version of SICD.  This means the SarPy code has to be able to handle FTITLE with or without it, in order to be completely generic.

https://nsgreg.nga.mil/doc/view?i=4901

SarPy should allow writing and reading of SICDs with the "SICD: " prefix in the FTITLE, but it should not force it.

johnhhenderson, 2021-03-17:
Thanks for the link.  Web searches keep taking me here: https://gwg.nga.mil/ntb/baseline/docs/SICD/.

https://nsgreg.nga.mil/doc/view?i=4901

utwade, 2021-03-17:
Yes, I'm aware.  Its unfortunate that we seem to be able to neither influence NGA or Google to move the actual current, approved spec higher in the search rankings. :(