ngageoint/sarpy #292: Reading  Spacenet 6 data from Capella Space Airborne testbed

wilsonjohnpatCO, 2021-12-13:
The Spacenet 6 data was from Capella Space's Airborne testbed.   The data is available on AWS S3.   Information on getting the data is available at 
https://medium.com/the-downlinq/spacenet-6-expanded-dataset-release-e1a7ddaf030

I attempted reading the data with the CapellaReader and this data is missing the field delta_line_time in the image_geometry

tommccullough-tenica, 2021-12-15:
Hello, thank you for your message. Unfortunately, that article references the original preliminary/test format for Capella data before they had even launched any satellites. The support for even the current Capella SLC format data is only partial. My efforts have to be carefully chosen, and I'm sorry that I can not justify the effort to support this out of date format. 

I will close this issue for now, but please feel free to respond to this thread if you sincerely believe this to be important enough to warrant more consideration.

tommccullough-tenica, 2021-12-15:
My manager

wilsonjohnpatCO, 2021-12-15:
Since writing the issue,  I had gone through and made an attempt.   I am not sure whether my changes are valid:
collect_id is not present,  so I set CoreName to file_name
delta_range_sample is not present, so I set SS to pixel_spacing_row
range_to_first_sample is not present,  so for r_ca_scp,  I retrieved the SCP and computed the range between the SCP and the evaluation of the ARP poly at the SCP time. 
since  first_line_time is not present,  I used the start_timestamp for first_time
for ss_zd_s, instead of delta_line_time,  I used pixel_spacing_column

This allowed to load the image.  I have not tried anything yet.   the remaining possible issues is that backprojection is not a valid SICD image formation algorithm so that is being set to "OTHER" and sigma_nought is not a recognized value for radiometry.

utwade, 2021-12-17:
Mostly looks OK at a high level.  A few words of caution:

Note that Capella and SICD terminology conventions swap the meanings of the terms "row" and "column".  A column to Capella is a row to SICD.  So delta_range_sample would map to Capella's pixel_spacing_column.

Also I understand that pixel_spacing_column is in ground plane, whereas delta_range_sample is in slant plane, so some care needs to be taken there.

Also be careful with your units.  ss_zd_s is in units of seconds, whereas pixel_spacing_row is in meters.

wilsonjohnpatCO, 2021-12-17:
The things you mentioned may be why trying to write a sicd or doing orthorectification is not working for me once I load an image with my hacked code.   

I will try to fix my hacked code next week, but after Christmas I will need to be working other projects for 3-4 weeks.

tommccullough-tenica, 2021-12-21:
Unfortunately the ability to use the sicd projection model (ortho-rectification requires this) is not going to work - this is part of the Capella SLC support being only partial at this point. There are outstanding questions about the Doppler Centroid polynomial, and a number of things depend very explicitly on this. I am hopeful that this will get resolved in the nearish term, but certainly not in the next week or two.

closed 2022-03-02 by tommccullough-tenica