ngageoint/sarpy #295: Issue with OrthorectificationIterator when creating SIDD from inmemory SICD

wilsonjohnpatCO, 2021-12-22:
I am attempting to create a SIDD from an inmemory SICD.   Creating a SIDD from an inmemory SICD seems to not work due to an issue in OrthorectificationIterator

new_meta=meta.copy()
    apod_shape=np.shape(tImage3a)
    new_meta.ImageData.NumRows=apod_shape[0]
    new_meta.ImageData.NumCols=apod_shape[1]
    new_reader=FlatSICDReader(new_meta, tImage3a)

    ortho_method=NearestNeighborMethod(new_reader, index=0, complex_valued=True, apply_radiometric=None, subtract_radiometric_noise=False)

  create_detected_image_sidd(ortho_method, '.', block_size=10, version=2, remap_function=High_Contrast())

I get the following error:
 File "/opt/miniconda3/lib/python3.7/site-packages/sarpy/io/product/sidd_product_creation.py", line 159, in create_detected_image_sidd
    remap_function=remap_function, recalc_remap_globals=False)
  File "/opt/miniconda3/lib/python3.7/site-packages/sarpy/processing/ortho_rectify.py", line 2294, in __init__
    if os.path.abspath(ortho_helper.reader.file_name) != \
  File "/opt/miniconda3/lib/python3.7/posixpath.py", line 378, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType

tommccullough-tenica, 2021-12-23:
Thanks for bringing this to my attention. I'll resolve this, and a couple of other issues with the ortho-rectification efforts hopefully next week.

tommccullough-tenica, 2021-12-29:
This should be resolved in version 1.2.46, thanks for your help

closed 2021-12-29 by tommccullough-tenica