ngageoint/sarpy #304: Does ground_to_image_geo work for SIDD products?

wilsonjohnpatCO, 2022-01-24:
I generated a SIDD image product from a SICD.   I attempted to convert a set of geodetic coordinates to image coordinates and got the following traceback:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/miniconda3/lib/python3.7/site-packages/sarpy/geometry/point_projection.py", line 1046, in ground_to_image_geo
    return ground_to_image(geodetic_to_ecf(coords, ordering=ordering), structure, **kwargs)
  File "/opt/miniconda3/lib/python3.7/site-packages/sarpy/geometry/point_projection.py", line 958, in ground_to_image
    coa_proj = _get_coa_projection(structure, use_structure_coa, **coa_args)
  File "/opt/miniconda3/lib/python3.7/site-packages/sarpy/geometry/point_projection.py", line 690, in _get_coa_projection
    if use_structure_coa and structure.coa_projection is not None:
AttributeError: 'tuple' object has no attribute 'coa_projection'

This occurs with both a SIDD produced by SARPY from the SICD and another processor.  
The ground_to_image_geo operation works correctly on the SICD,  producing image coordinates in slant plane.

wilsonjohnpatCO, 2022-01-24:
I tried processing another SICD from another platform to SIDD using sarpy and then converting the center coordinates of the scene to image coordinates and got the same error.

tommccullough-tenica, 2022-01-25:
Hello, many thanks for your issue. You didn't quite provide enough context for your actual function call. I infer from the error that what you passed in from `structure` is a tuple, instead of a single SICD or SIDD structure.

I would bet that you likely did something along the lines of `reader = SIDDReader(<path>)`, and then passed in the contents of `reader.sidd_meta` for the structure argument, which is likely returning a tuple in this case (might be a tuple with a single element). See the documentation [here](https://sarpy.readthedocs.io/en/latest/io/product/base.html). 

I recommend that you use `reader.get_sidds_as_tuple()[index]` to avoid this annoyance. I recommend the same approach for sicd-type readers (`reader.get_scids_as_tuple()[index]`) too, unless you only read literal SICD files. I'll leave this issue open for a day or two, but will close it if I don't hear back from you.

wilsonjohnpatCO, 2022-01-25:
That was the issue.  It was not apparent from the example of using the point_projection methods that one needed to use get_sidds_as_tuple rather than sidd_meta.

closed 2022-01-26 by tommccullough-tenica