ngageoint/sarpy #315: Writing large SICDs does not work

kjurka, 2022-03-01:
When trying to write out a large SICD, I get an error.  Using the same test code from: https://github.com/ngageoint/sarpy/issues/233

I run the below to generate a ~12GB SICD to force it to segment:

```
$ ~/a3-2019-10-rn/bin/python make_multi_seg_image.py ~/sample-sicds/sicd_example_1_PFA_RE32F_IM32F_HH.nitf big-multi-seg.nitf 200100 8000
```

and it doesn't like that.

```
Traceback (most recent call last):
  File "make_multi_seg_image.py", line 28, in <module>
    writer = sicd.SICDWriter(args.OutFile, meta)
  File "/home/kjurka/git/kjurka/sarpy/sarpy/io/complex/sicd.py", line 638, in __init__
    super(SICDWriter, self).__init__(file_name, check_existence=check_existence)
  File "/home/kjurka/git/kjurka/sarpy/sarpy/io/general/nitf.py", line 2196, in __init__
    self._create_image_segment_details()
  File "/home/kjurka/git/kjurka/sarpy/sarpy/io/complex/sicd.py", line 781, in _create_image_segment_details
    pixel_size, dtype, transform_data, pv_type, isubcat, image_segment_limits = self._image_parameters()
  File "/home/kjurka/git/kjurka/sarpy/sarpy/io/complex/sicd.py", line 775, in _image_parameters
    self.sicd_meta.ImageData.NumRows, self.sicd_meta.ImageData.NumCols, pixel_size)
  File "/home/kjurka/git/kjurka/sarpy/sarpy/io/general/nitf.py", line 2122, in image_segmentation
    'and this segmentation will require row portion of ILOC to be {}'.format(row_offset))
ValueError: NITF image segment ILOC field is limited to 5 digits in each of row/column,
        and this segmentation will require row portion of ILOC to be 156249
```

kjurka, 2022-03-01:
OK, it does work if you don't end up with a number of rows > 99999.  It's odd to be able to create a 200000 row SICD as long as it's super narrow (not segmented), but if it does get segmented, then you can't have that many rows.  That seems kind of backwards as segmenting is exactly the solution to large imagery...

tommccullough-tenica, 2022-03-02:
Clearly this is a bug, bear with me

tommccullough-tenica, 2022-03-02:
Okay, I'm thinking clearly about this now, and the answer is not satisfying (to me anyways).

The SICD standard requires that images are segmented in only row space - that is, each image segment is the entire swatch of columns for an appropriate number of rows (so the image segment size doesn't exceed the total allowed image size in bytes).

For each image segment, the relative index of the minimum row/column corner (upper left in regular image visualization) for this image segment as a part of the total image is provided in NITF Image Segment Subheader ILOC field, which is limited to 5 digits for each direction (because the NITF standard is both old and old-fashioned). 

The upshot is that I see no valid way to make a SICD (or SIDD, which specifies the same segmentation rules) where the segmentation rules indicate a segment which starts at row 100,000. Although the SICD segmentation scheme is really not robust, the fundamental shortcoming is still the size constraint of the NITF Image Segment ILOC field.

The best that I can do for now is to raise a more informative exception. It obvious that the SICD standard needs to be refined to permit things like better cloud reading behavior and the ability to make larger images.

bombaci-vsc, 2022-03-02:
A given image segment's ILOC is relative to the image segment it's attached to (specified by IALVL).  For a SICD, the ILOC rows need to be set to the number of rows of the *previous* segment, not the "global" location.  

It looks like the SICD writer is setting ILOCRow to the current segment's row offset, not the number of rows in the previous segment:
https://github.com/ngageoint/sarpy/blob/8619f4dfd269e8598e73b219261282ff24ea5b45/sarpy/io/complex/sicd.py#L823

Take a look at SICD Volume 2 Section 3.2.3 Example 2.  There you'll see the ILOCRow values for a 3 segment case.

kjurka, 2022-03-02:
Also, the computation of the `row_count` is not capped at 99999

https://github.com/ngageoint/sarpy/blob/a2a413a230aaa700488a2f200e68f4a38bbcfd21/sarpy/io/general/nitf.py#L2117

If it exceeds that value, you have a problem.

tommccullough-tenica, 2022-03-02:
Nathan, that was my reading of the NITF standard, so that is most certainly what I was writing. I guess what I am writing is appropriate for the unattached version, and not the attached version. Many thanks for the trouble shooting.

closed 2022-03-02 by tommccullough-tenica