ngageoint/sarpy #340: Error validating signal write data

RyanTerb, 2022-08-11:
Receiving the following error writing a CPHDWriter1 file:
 
`sarpy.io.phase_history.cphd::ERROR - signal data at index 0 not written`

Signal data ([write_signal_block](https://github.com/ngageoint/sarpy/blob/64a7dc28477744964da150f0a1c2479cb859d579/sarpy/io/phase_history/cphd.py#L1698)) is successfully written and I can read it back in. The code doesn't seem to update the `details.item_written = True` for signal block data.

Both [PVP ](https://github.com/ngageoint/sarpy/blob/64a7dc28477744964da150f0a1c2479cb859d579/sarpy/io/phase_history/cphd.py#L1625) and [Support](https://github.com/ngageoint/sarpy/blob/64a7dc28477744964da150f0a1c2479cb859d579/sarpy/io/phase_history/cphd.py#L1592) have this update.

Is this an oversight? or do I need to make an extra call somewhere?

thomasmccullough, 2022-08-11:
Hi Ryan(?), thanks for the bug report. I did a refactor which made that check obsolete, but failed to remove the check. Everything passed the unit/integration tests, and I just failed to see the logging. This should be fixed shortly.  -Tom

closed 2022-08-11 by thomasmccullough

RyanTerb, 2022-08-11:
Thanks @thomasmccullough, verified that the logging error I was seeing is gone with the latest push.