ngageoint/sarpy #357: Attribute Error with populate_nitf_information_into_sicd()

jbolt256, 2022-11-16:
Tested on Python 3.7.10, with sarpy 1.3.9. If this has been fixed since 1.3.9, I apologize but I do not have access to any newer version.

Example code:
```
from sarpy.io.complex.sicd import SICDReader, SICDWriter
reader = SICDReader("the_sicd_file.sicd")
reader.populate_nitf_information_into_sicd()
```

Provides the following error:
---------------------------------------------------------------------------
```
AttributeError                            Traceback (most recent call last)
/tmp/ipykernel_4088966/3690750003.py in <module>
      1 from sarpy.io.complex.sicd import SICDReader, SICDWriter
      2 reader = SICDReader("the_sicd_file.sicd")
----> 3 reader.populate_nitf_information_into_sicd()

.../lib/python3.7/site-packages/sarpy/io/complex/sicd.py in populate_nitf_information_into_sicd(self)
    339         """
    340 
--> 341         self._sicd_meta.NITF = self.get_nitf_dict()
    342 
    343     def depopulate_nitf_information(self):

.../lib/python3.7/site-packages/sarpy/io/xml/base.py in __setattr__(self, key, value)    714                 'Class {} instance receiving unexpected attribute {}.\n\t'
    715                 'Ensure that this is not a typo of an expected field name.'.format(self.__class__.__name__, key))
--> 716         object.__setattr__(self, key, value)
    717 
    718     def __getstate__(self):

AttributeError: can't set attribute

```
The issue seems to originate here: https://github.com/ngageoint/sarpy/blob/master/sarpy/io/complex/sicd_elements/SICD.py#L224
There is no setter for NITF(). Simply adding a basic setter below the @property definition for NITF() in SICD.py seems solves the issue:
```
    @NITF.setter
    def NITF(self, value):
        self._NITF = value
```

Darder, 2022-11-17:
Thank you for the feedback, will look into this issue.

Darder, 2022-11-22:
pull request #358 should alleviate this issue. Thank you.

closed 2022-11-22 by Darder