ngageoint/sarpy #359: Stale Links

RyanTerb, 2022-12-09:
Not super important, but the links to the CPHD / SICD / SIDD specs in the README are no longer valid.

Darder, 2022-12-12:
What version of SARPY are you using? 

Links to CPHD, SICD, and SIDD specs were updated 20 days ago with version 1.3.12

RyanTerb, 2022-12-12:
@utwade Thanks, I'll keep checking the links as well.

joshday, 2023-01-22:
Has a mirror for the links been found?  I have been searching for Volume 1 with no luck.  If anyone reading this has a local PDF copy, I would be very grateful if you could share it with me.

utwade, 2023-01-23:
[NGA.STND.0024-1_1.3.0_SICD_DIDD_FINAL.pdf](https://github.com/ngageoint/sarpy/files/10484454/NGA.STND.0024-1_1.3.0_SICD_DIDD_FINAL.pdf)

Let me see if I can just drop the current SICD vol 1 into this issue conversation, since the primary website is still not back up.

utwade, 2023-02-22:
Links back online.  Closing issue.

closed 2023-02-22 by utwade