ngageoint/sarpy #371: File Not Found error - SentinelReader

jayepraveen999, 2023-02-12:
Hi everyone!

I came across the package Sarpy since I want to retrieve the complex values of a particular pixel of Sentinel 1 SLC data using python.
I have used SentinelReader (from sarpy.io.complex.sentinel import SentinelReader) to read the downloaded SLC data (.SAFE file) but I am facing File not Found error even though the path provided is correct and the file exists in the path.
Attached are the screenshots of this issue.

Sorry for asking you some basic questions but I couldn't figure out this error. 
[image]
[image]

utwade, 2023-02-13:
The first and last pixels in Sentinel SLC are usually zeroes (which are what are being displayed in the attached screenshot), since there are black bars on the top and bottom of the image.  Have you tried looking in some of the middle pixels?

With regard to the radiometric parameters, the SAFE directory structure should have an "annotation" directory and then a "calibration" subdirectory under that.  Within that, there should be calibration* and noise* files for each measurement GEOTIFF.  Can you check to make sure that structure exist like that?

utwade, 2023-02-17:
Yes, you definitely have to unzip.

If you are still having this issue, is there any chance you put the directory structure (with measurement directory deleted) back into a zip and then attach it to this issue conversation for us to review?  We don't see the same issue with the radiometric parameters.  Might be something we could tell at a glance if we could see your directory structure.

closed 2023-05-03 by Darder