ngageoint/sarpy #404: Unclosed file object

RyanTerb, 2023-05-02:
Noticed with sarpy 1.3.30 that the sicd_writer is throwing a warning:

```sh
/usr/local/lib/python3.10/site-packages/sarpy/io/general/nitf.py:4168: ResourceWarning: unclosed file <_io.BufferedWriter name=
```

It looks like a file object is not being close properly and python is unhappy.  Based on the this [code](https://github.com/ngageoint/sarpy/blob/8fb2caea6b417da26675d22865c798f3220b0591/sarpy/io/general/nitf.py#L4178), it might not be triggering the ```__del__``` operator that closes the file object, [link](https://github.com/ngageoint/sarpy/blob/8fb2caea6b417da26675d22865c798f3220b0591/sarpy/io/general/nitf.py#L1236).

The following code removed the warning for me. I had to explicitly call flush though.

```python
  writer = sicd.SICDWriter(
      file_object=filename,
      sicd_meta=SICDType(**sicd_data.__dict__),
      check_existence=False
  )

  writer.write(complex_image, (0, 0))
  writer.flush()
  writer._file_object.close()
  writer.close()
```

If I'm incorrectly writing a sicd_file or closing, please advise.

Thanks !

Darder, 2023-05-03:
Did you not run into this problem before with earlier versions of SARPY?

RyanTerb, 2023-05-08:
I have not.

pressler-vsc, 2023-11-10:
@RyanTerb, thanks for writing this up. This does seem to be a bug in the NITFWriter introduced in the 1.3.30 refactor and is also an issue with the CPHDWriter. This should be addressed in #476.

bombaci-vsc, 2024-10-04:
Included in version 1.3.59

PR #476 (2023-11-13): Unclosed file objects in NITF/CPHD writer close methods
# Description
Closes #404

This MR addresses #404 by adding a missing `_file_object.close()` to the `close` methods of the NITFWriter and CPHDWriter1.

I originally intended to keep the `_file_object = None` behavior intact, however when I did so, I found that the `close` methods were being called twice: the first time was explicitly in the code (or implicit in the case of a context manager) and a second time in a `__del__` method. We _could_ code around this, but nothing seemed to rely on this behavior and it seemed unecessary given that [... cut]
closed 2024-10-04 by bombaci-vsc