ngageoint/sarpy #442: Default AmpTable for AMP8I_PHS8I

magelisk, 2023-07-05:
Currently, if SICD is provided with pixel type `AMP8I_PHS8I`, [the code](https://github.com/ngageoint/sarpy/blob/master/sarpy/io/complex/sicd.py#L361C55-L362) requires that `AmpTable` be provided. I believe that the code should provide a default AmpTable of just 0-255 instead of raiding a `ValueError`. SICD spec says:

> If no amplitude look-up table is provided, pixel component AMP(row,col) is the pixel amplitude. Pixel amplitude AII(row,col) is determined as shown below.
> AMP(row,col), if AmpTable(0:255) is omitted

dhaverporth, 2025-05-06:
I believe this is a problem with the SICD schema that is published with the standard rather than a SARPy bug per say.  You quoted the text in the SICD spec but the schema says that if AMP8I_PHS8I is used that an AmpTable must be present.  If a required field (as dictated by the schema) is not present a ValueError is raised.  I recommend just filling out a basic lookup table in your SICD to fix the error.  The SARPy team has no real authority to changed the SICD schema.

pressler-vsc, 2025-05-06:
@dhaverporth, can you share the schema you're referencing? I can't find such a constraint in those from https://nsgreg.nga.mil/

e.g.:
https://github.com/ngageoint/sarpy/blob/bb17d79315ca5fead3da4db11013fd1430ed0f6a/sarpy/io/complex/sicd_schema/SICD_schema_V1.2.1_2018_12_13.xsd#L360-L376

```python
# start with a valid XML instance
>>> import lxml.etree
>>> sicd_xml = lxml.etree.parse("tests/data/example.sicd.xml")
>>> SCHEMA = lxml.etree.XMLSchema(file="sarpy/io/complex/sicd_schema/SICD_schema_V1.2.1_2018_12_13.xsd")
>>> print(f"{SCHEMA.validate(sicd_xml)=}")
SCHEMA.validate(sicd_xml)=True

# change PixelType and schema is still valid
>>> sicd_xml.find(".//{*}PixelType").text = "AMP8I_PHS8I"
>>> print(f"{sicd_xml.findtext(".//{*}PixelType")=}")
sicd_xml.findtext(".//{*}PixelType")='AMP8I_PHS8I'
>>> print(f"{sicd_xml.find(".//{*}AmpTable")=}")
sicd_xml.find(".//{*}AmpTable")=None
>>> print(f"{SCHEMA.validate(sicd_xml)=}")
SCHEMA.validate(sicd_xml)=True
```

Reading/writing such a SICD is possible with [SARkit](https://github.com/ValkyrieSystems/sarkit):
```python
import sarkit.sicd as sksicd
sec = sksicd.NitfSecurityFields(clas="U")
meta = sksicd.NitfMetadata(
    xmltree=sicd_xml,
    file_header_part=sksicd.NitfFileHeaderPart(ostaid="my station", security=sec),
    im_subheader_part=sksicd.NitfImSubheaderPart(isorce="my sensor", security=sec),
    de_subheader_part=sksicd.NitfDeSubheaderPart(security=sec),

)
with open("amp8i_phs8i_nolut.sicd", "wb") as f, sksicd.NitfWriter(f, meta) as w:
    pass # don't care about pixels

with open("amp8i_phs8i_nolut.sicd", "rb") as f, sksicd.NitfReader(f) as r:
    img = r.read_image()
    print(f"{img.dtype=}, {img.shape=}")
    print(f'{r.metadata.xmltree.find(".//{*}AmpTable")=}')
```

```
img.dtype=dtype([('amp', 'u1'), ('phase', 'u1')]), img.shape=(1494, 1723)
r.metadata.xmltree.find(".//{*}AmpTable")=None
```

petersontex, 2025-05-06:
> I believe this is a problem with the SICD schema that is published with the standard rather than a SARPy bug per say. You quoted the text in the SICD spec but the schema says that if AMP8I_PHS8I is [...]

Closing this issue per @dhaverporth's comment.

closed 2025-05-06 by petersontex