ngageoint/sarpy #466: register_arrays typo

HGS-JoeDecker, 2023-08-22:
I am trying to register two arrays and have encountered the following.

It appears that on like 231 in regi.py moving_array should be moivng_array.ndim ?

>>> result = register_arrays(data1, data2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\jdecke04\AppData\Local\Programs\Python\Python310\lib\site-packages\sarpy\processing\registration\regi.py", line 547, in register_arrays
    return _single_step_grid(
  File "C:\Users\jdecke04\AppData\Local\Programs\Python\Python310\lib\site-packages\sarpy\processing\registration\regi.py", line 487, in _single_step_grid
    best_loc, max_correlation = _single_step_location(
  File "C:\Users\jdecke04\AppData\Local\Programs\Python\Python310\lib\site-packages\sarpy\processing\registration\regi.py", line 373, in _single_step_location
    best_temp_location, maximum_correlation = _max_correlation_step(reference_array, moving_array, do_subpixel=do_subpixel)
  File "C:\Users\jdecke04\AppData\Local\Programs\Python\Python310\lib\site-packages\sarpy\processing\registration\regi.py", line 231, in _max_correlation_step
    if reference_array.ndim != 2 or moving_array != 2:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

petersontex, 2025-05-06:
Hello @HGS-JoeDecker, 

Did you resolve this issue? Do you still require assistance?

petersontex, 2025-05-12:
No response from end user. Closing this issue.

closed 2025-05-12 by petersontex