ngageoint/sarpy #474: Support reading CF8 CPHDs with an AmpSF PVP

pressler-vsc, 2023-10-27:
SarPy's CPHD reader currently errors out when attempting to open a CF8-valued CPHD that contains an AmpSF PVP:

```python
>>> import sarpy.io.phase_history.converter
>>> sarpy.io.phase_history.converter.open_phase_history('/data/cf8_with_ampsf.cphd')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/converter.py", line 79, in open_phase_history
    reader = opener(file_name)
             ^^^^^^^^^^^^^^^^^
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/cphd.py", line 1011, in is_a
    return CPHDReader(cphd_details)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/cphd.py", line 519, in __init__
    data_segments = self._create_data_segments()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/cphd.py", line 564, in _create_data_segments
    format_function = AmpScalingFunction(raw_dtype, amplitude_scaling=amp_sf)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/cphd.py", line 78, in __init__
    self.set_amplitude_scaling(amplitude_scaling)
  File "/home/vscuser/git/glweb/software/third-party/sarpy/sarpy/io/phase_history/cphd.py", line 126, in set_amplitude_scaling
    raise ValueError(
ValueError: A scaling multiplier has been supplied,
        but the raw datatype is not `int8` or `int16`.
```

This [documented limitation](https://github.com/ngageoint/sarpy/blob/11d19a0440725181678cff89873c6f2315c6a96d/sarpy/io/phase_history/cphd.py#L126-L128) is not a limitation of the CPHD spec (e.g. from NGA.STND.0068-1_1.1.0_CPHD, pg. 25):

[image]

SarPy should support reading these CPHDs.

eleehiga, 2023-11-15:
I as well would please want SarPy to support this

bombaci-vsc, 2024-10-04:
Included in version 1.3.59

closed 2024-10-04 by bombaci-vsc