ngageoint/sarpy #475: Request for support of Sandia's `FARAD SAR DATA COLLECTION` `Complex SAR Data` format

eleehiga, 2023-11-08:
Trying to read a complex SAR data file similar to:
```python
from sarpy.io.complex.converter import open_complex

reader = open_complex('FARAD_X_BAND/20151027_1027X04_PS0010_PT000001_N03_M1_CH0_OSAPF.ntf ')
```
This yields the error `sarpy.io.general.base.SarpyIOError: Unable to determine complex image format.` The data is from here: https://www.sandia.gov/files/radar/complex-data/FARAD_X_BAND.zip and from this website here: https://www.sandia.gov/radar/pathfinder-radar-isr-and-synthetic-aperture-radar-sar-systems/complex-data/. Have tried the GFF reader, as this data is from Sandia, here: https://sarpy.readthedocs.io/en/latest/io/complex/gff.html but that still led to errors. Therefore was wondering if you guys could please support Sandia's `FARAD SAR DATA COLLECTION` `Complex SAR Data` format and the beginning of this file is here:
```
NITF02.1005BF01SNLSAR    2017112220271820151027_1027X04_PS0010_PT000001_N03_M1_CH0_OSAPF.ntf                           U                                                                                                                                                                      00000000000                                             00009242013100040400100129500924184320000000000000000000000000IM000000000120151027204130                 Sandia Complex                                                                  U                                                                                                                                                                      0OSAPF                                     0000402800005736INTMULTI   SAR     16RU13SCU583529160213SCU585339123413SCU582169107913SCU58036914460NC2M M     N   0  P     N   00S00010001573640281600100000000000001.0 0000000783000ACFTB 002071027X04                                   SAR SNLSAR00000012015102700000000000004+35.15500303-106.55589243000.00+01646m+00.00000000+000.00000000      090.00000.0770m00.0714m999.990000010008.22        0001000AIMIDB0008920151027204130UNKN1027X04   00000AA00000 00100001AA00100001US    3509N10633W             BLOCKA00123010000004028090270                +35.154081-106.553146+35.152637-106.556598+35.155926-106.558638+35.157370-106.555187010.0EXPLTB00101332.99900.000+00.00000.000UNK                28.5800.0028.58HH0573601                           00000MENSRB00205+35.13885676-106.59445361000000012462+35.15500303-106.55589243000000+05400+0000.0+0000.00.878190014755L0201402868-0.3986985-0.7824651-0.4783178-0.891001+0.454002+0.000000+0.217157+0.426182-0.87818700100001 " &  ! 2 = = " +  5  &   D     , &  9 <     4  T  P  2 ' 8   #     
     '  5  (  	 )      L  > 2 L  + O 2 r b k > I  =  3  $ 
```

eleehiga, 2023-11-14:
This file is readable with MATLAB with this reader here: https://github.com/ngageoint/MATLAB_SAR/blob/master/IO/complex/nitf/open_cnitf_reader.m. Therefore was wondering if you guys could please implement this reader in sarpy.

dhaverporth, 2025-05-06:
This is an old data format not supported by SARPy.  SARPy is tasked with reading NGA approved formats.

petersontex, 2025-05-07:
> This is an old data format not supported by SARPy. SARPy is tasked with reading NGA approved formats.

Closing this issue per @dhaverporth's comment.

closed 2025-05-07 by petersontex