ngageoint/sarpy #478: Request for Sarpy to support CPHD from Gotcha Volumetric SAR Data Set Overview: https://www.sdms.afrl.af.mil/index.php?collection=gotcha

eleehiga, 2023-11-15:


fkoehleriv, 2024-04-07:
When the Gotcha page you referenced mentions 'SAR phase history', I believe they're referring to SLC (single look complex) that's already formed to the image domain (using about 1 degree of aperture), not raw (video) or compensated phase history (unfortunately).  That's because the description of the contents of the MAT file notes K-space data.  Rather than converting this to CPHD, it may be helpful to be able to read the MAT files and create a rough SICD.  This might enable easier experiments with interferometry & 3D SAR point cloud creation, etc.  So... I agree with your request to add support for Gotcha, but it would be converting the Gotcha wideband and narrowband data in their MAT files into a SICD...not a CPHD.

rachelglenn, 2025-01-23:
Hi. Was this enhancement ever added?

petersontex, 2025-05-07:
@dhaverporth should we add this enhancement to our planning for future releases?