ngageoint/sarpy #482: How to apply image formation to CPHD

eleehiga, 2023-11-20:
I followed the example to read in CPHD here: https://sarpy.readthedocs.io/en/latest/examples/reading.html#phase-history-cphd-readers but was wondering how to apply image formation to this in order to generate an image?

johnstob, 2023-11-27:
Unfortunately, `sarpy` does not implement an image formation processor.  You might have a look at [MATLAB_SAR Toolbox](https://github.com/ngageoint/MATLAB_SAR/tree/master/Processing/IFP)  and see if it suits your needs.

utwade, 2023-11-28:
That is correct.  If you don't mind MATLAB, the MATLAB SAR Toolbox does include simple, academic implementations of polar formatting and backprojection for spotlight CPHD.

closed 2023-11-28 by utwade