ngageoint/sarpy #515: Unable to validate CPHD data because parse_xml_from_string returns xml_ns=None

dberwick, 2024-05-31:
Have been unable to read some recently received CPHD files. I believe they are version 1.1.0. Admittedly I don't understand enough about XML namespaces. However, in base.py there is an attempt to Iterparse the XML extracted from the CPHD. There seems to be no start-ns events generated. Does this imply malformed CPHD? Seems like there is code in `parse_xml_from_string` to address this case by checking the length, but if `xml_ns = None` then I think `CPHDType::from_xml_string` errors out looking for 'default' in None.

pressler-vsc, 2024-05-31:
It is difficult to say without seeing the file, but it does sound like the XML could be missing a namespace declaration. The version-specific CPHD namespaces are defined in the accompanying schema files (e.g. [nsgreg link for CPHD 1.1.0 schema](https://nsgreg.nga.mil/doc/view?i=5421)):

```xml
<?xml version="1.0" encoding="UTF-8"?>
<xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema"
   elementFormDefault="qualified"
   attributeFormDefault="unqualified"
   targetNamespace="http://api.nsgreg.nga.mil/schema/cphd/1.1.0"        <----- this
   xmlns="http://api.nsgreg.nga.mil/schema/cphd/1.1.0">
...
```

There are a few ways to [declare namespaces in XML 1.0](https://www.w3.org/TR/xml-names/#ns-decl). Perhaps the simplest is to specify the default namespace in the root element, as can be seen in this [syntax-only test XML file](https://github.com/ngageoint/sarpy/blob/master/tests/data/syntax-only-cphd-1.1.0-monostatic.xml#L1)

```xml
<CPHD xmlns="http://api.nsgreg.nga.mil/schema/cphd/1.1.0">
...
```

As noted in NGA.STND.0068-1_V1.1.0 _CPHD Section 2.1:  

  > The two-byte section terminator “\f\n” (0x0C, 0x0A) marks the end of the File Header. The section terminator enables a reader application to identify the end of the File Header block and enables viewing of the file header using standard functions on some systems (e.g., the “more” command on a UNIX system).

This property makes it possible to inspect the version of the CPHD file as well as to navigate to the portion of the file that contains the XML block using readily available tooling, e.g.

  ```shell
  $ more /data/bistatic_spotlight_example.cphd | head -n 3
  CPHD/1.1.0
  XML_BLOCK_SIZE := 24846
  XML_BLOCK_BYTE_OFFSET := 1024
  ```
  ```shell
  $ tail -c +1024 /data/bistatic_spotlight_example.cphd | head -c 100
  <CPHD xmlns="http://api.nsgreg.nga.mil/schema/cphd/1.1.0"><CollectionID><CollectorName>SyntheticCol
  ```

Hopefully this gets you closer to resolving the issue.

dberwick, 2024-06-03:
Thanks for the quick response. Using the head/tail commands provided, clear (now) that the namespace schema was not specified in the XML. The version seems to be 1.1.0, so... I guess I can try to hack the schema in and see if that gets further. 

FWIW I altered CPHD.py to test `xml_ns` before looking for 'default'. 
```
@classmethod
    def from_xml_string(cls, xml_string):
        ...
        root_node, xml_ns = parse_xml_from_string(xml_string)
        if xml_ns == None:
            ns_key = None
        else:
            ns_key = 'default' if 'default' in xml_ns else None
        return cls.from_node(root_node, xml_ns=xml_ns, ns_key=ns_key)
```

Makes it a further, but then dies trying to parse the `CODTimePoly` and `DwellTimePoly`. I only mention this because it does seem like `parse_xml_from_string` can legitimately return a `None` and the error check here might be necessary. 

I'm now afraid that there are issues beyond the missing namespace schema, but as I mentioned will try to force the schema for now and see if that helps.

dberwick, 2024-06-03:
Yup. Still dies trying to convert `CODTImePoly` and `DwellTimePoly` even if I force the namespace to match the 1.1.0 standard. I'll see if I can work that a bit. 

From my perspective can close the ticket unless you want to consider the error handling adjustment above.

pressler-vsc, 2024-06-05:
Glad to hear you're a bit closer and thanks for the note. We agree that there are some odd edge cases with respect to the XML parsing/handling as you've pointed out. We'll likely leave this as is and only fix critical bugs (reading valid files) as we're working on the next major release of SarPy that has a much lighter-weight metadata model.

If you encounter any further issues, let us know.

closed 2024-06-05 by pressler-vsc