ngageoint/sarpy #517: Not clear how to write .sio file - SIOWriter()

shckltn, 2024-06-10:
Firstly, sorry if this is the wrong place to post this.

I've been bashing my head against trying to figure out how to write an .sio file using SIOWriter(), but I can't seem to find decent example code anywhere. If anyone has a good example of writing a data set to and .sio file using sarpy please let me know.

It seems that many sicd meta data parameters need to be filled before being able to use the SIOwriter() function with the following function signature:
class sarpy.io.complex.sio.SIOWriter(file_object: str | BinaryIO, sicd_meta: SICDType, user_data: Dict[str, str] | None = None, check_older_version: bool = False, check_existence: bool = True)

Trying to initialize a new SICDType

pressler-vsc, 2024-08-26:
SARPy's SIO writer is not intended to support generic SIO I/O, but only in the context of an SIO with metadata sufficient to populate a SICD structure. If this is still applicable, it is possible nga-six [https://github.com/ngageoint/six-library](https://github.com/ngageoint/six-library) may have such a capability.

pressler-vsc, 2024-08-27:
@shckltn, I think I misread your question and wanted to clarify my prior comment. As you noted, SARPy's SIOWriter is [currently hardcoded](https://github.com/ngageoint/sarpy/blob/3354207e6e8d8920416d1248c9c626c7d5a86b47/sarpy/io/complex/sio.py#L449C9-L449C18) to write SIOs with user-data containing SICD XML and does not support the metadata-free format that the [MATLAB_SAR toolbox does](https://github.com/ngageoint/MATLAB_SAR/blob/master/IO/complex/sio/write_sio.m).

If SIOs with user-data are acceptable, then what you are attempting _should_ be possible with SARPy, however there are a few issues getting in the way (assuming you are using the latest release):
- the error you are receiving is due to an incorrect argument while instantiating a `SIOWriter`; the third positional argument is `user_data` which should be a dict of str -> str for use when adding userdata (beyond the SICD XML) to the output SIO. If you don't have such userdata, you can remove that arg, e.g. `sioWriter = SIOWriter(output_file, sicd_meta)`
- The bigger issue: while trying to add this case in the form of a unittest, we encountered a number of bugs in `sarpy/io/complex/sio.py` that prevented writing. We plan to address these in a PR shortly.

Thanks for highlighting this issue!

bombaci-vsc, 2024-10-04:
Included in version 1.3.59

closed 2024-10-04 by bombaci-vsc