ngageoint/sarpy #523: Support for S3 bucket

effeminati, 2024-07-08:
I would like to process data directly on the cloud without the need to download data to the local filesystem. I tried, but I was not able to open remote files with Sarpy.

Do you foresee extending the interface to do this?

Sarpy is a valuable library capable of opening and standardizing data from many different SAR sensors. Making it able to work on the cloud would make a significant difference.

pressler-vsc, 2024-07-08:
Yes! As you point out, only a very limited subset of interfaces [(e.g.)](https://github.com/ngageoint/sarpy/blob/3354207e6e8d8920416d1248c9c626c7d5a86b47/tests/io/complex/test_remote.py#L22) are architected in a way that is compatible with something like smart-open.

There is currently an effort underway to modernize SarPy, and standardizing I/O in a way that enables working with large remote files is a key goal.

naething, 2024-07-25:
For the file types that are currently supported you can see an example of how to do so here:

https://sarpy.readthedocs.io/en/latest/examples/first_example.html#opening-remote-file

petersontex, 2025-06-18:
@effeminati thank you for your request. Per the above comments this capability is in sarpy.smart_open. Please let us know if you are having any issues with smart_open.

closed 2025-06-18 by petersontex