ngageoint/sarpy #528: cphd_consistency.py PVP bug

vaughn-holmes, 2024-07-18:
The `cphd_consistency.py` script under `sarpy.consistency` does not have logic to handle the `TxAntenna` and `RcvAntena` PVP fields properly and crashes. 

Currently the code to parse all the PVP elements is:
```python
    if elem.tag == "AddedPVP":
        name = elem.find('Name').text
    else:
        name = elem.tag

    offset = int(elem.find('Offset').text)
    size = int(elem.find('Size').text)

    dtype = cphd1_utils.binary_format_string_to_dtype(elem.find('Format').text)

    return name, {"offset": offset,
                  "size": size,
                  "dtype": dtype}
```
Since the antenna PVP fields specify a set of values and are not in themselves a PVP value, they do not have associated offset, size, and dtype.
[image]

vaughn-holmes, 2024-07-18:
UPDATE: I see on the master branch there's a fix for this. When is the next release of sarpy scheduled? It has been almost a year and there are several major updates that have gone in.

bombaci-vsc, 2024-07-18:
The content for the next version of sapry is being integrated in the `integration/1.3.59-rc` branch.  There are some challenges to overcome before it can become a formal release.

It looks like [a fix for this was made](https://github.com/ngageoint/sarpy/commit/2634acd869818e564fdc1645894b9e32ba439113#diff-40594572e8169b144e447ae57d6c93a6aaf047a43dd46de34a46f592874e9c67R282) and was released in sarpy 1.3.55 and looks to still be present in 1.3.58.  Both of which are available on pypi.   

Does this fix not handle your use case? or is there another package repository you are using?

vaughn-holmes, 2024-07-19:
Interesting. I thought I tested a 'clean' installation of the latest version yesterday, but something with my other dependencies must have still given me the older version we depend on (`1.3.24`). I just forced an install of the latest release and confirmed that it does work, thank you!

naething, 2024-07-25:
It looks like this fix is already present, and with the forced install of the latest release you aren't having any problems, so I am going to close this issue.

closed 2024-07-25 by naething