ngageoint/sarpy #551: Numpy 2 incompatibility for cphd.py and sicd.py: numpy.cast was removed from Numpy 2.1; suggest replacing with numpy.astype

jdgorman8r, 2024-10-01:
sarpy 1.3.58 contains numpy.cast in both sicd.py and cphd.py.  Running sarpy with numpy 2.1 results in errors when using readsicd or readcphd functions.  Suggest replacing numpy.cast() with numpy.astype() as follows:

 sicd.py, line 95:
      lookup_table = numpy.cast['float32'](lookup_table)
  
becomes: 
      lookup_table = numpy.astype(lookup_table, dtype='float32') or
           lookup_table = lookup_table.astype('float32')

cphd.py line 122:
       array = numpy.cast['float32'](array)
becomes:
      array = numpy.astype(array, dtype='float32') or
           array = array.astype('float32')

pressler-vsc, 2024-10-01:
@jdgorman8r, thanks for noting this. Since 1.3.58, development has been performed in the [`integration/1.3.59-rc` branch](https://github.com/ngageoint/sarpy/tree/integration/1.3.59-rc). You can read a bit more about integration branches [here](https://github.com/ngageoint/sarpy/tree/master#integration-branches).

numpy 2 compatiblity has come up in at least two PRs:
- #522 
- #535 

I believe the former addresses the specific lines of code you address. If you are able to test using `1.3.59-rc`, please let us know if you encounter any further incompatiblities.

jdgorman8r, 2024-10-01:
Thanks Daniel. We can close this issue.  Appreciate the quick response!

closed 2024-10-01 by jdgorman8r