ngageoint/sarpy #562: AMP8I_PHS8I AmpTable Overflow

shiozawt, 2024-10-21:
The file `format_function.py`, located at `sarpy/io/general/format_function.py` contains a bug within the function `_forward_magnitude_theta()` relating to the conversion from mag/phase to complex values. When passing in values of type `uint8`, the operation on line 780 where theta is multiplied by 2 * pi / (1 << bit_depth) leads to a memory overflow error, wrapping the output `uint8` stored values and putting the output complex value into the wrong plane in many cases. Recommended that a fix such as specifying 2.0 rather than ambiguous 2  on line 780.

For example:
A theta value of uint8(130) here wraps to '4' rather than the desired '260'.

bombaci-vsc, 2024-10-21:
A fix for this went in with the release of v1.3.59.  Could you check if you're still experiencing the problem using the latest sarpy?

https://github.com/ngageoint/sarpy/blob/master/sarpy/io/general/format_function.py#L780

shiozawt, 2024-10-21:
That is indeed fixed in a newer version. Thanks! Closing.

closed 2024-10-21 by shiozawt