ngageoint/sarpy #563: Capella SLC tiff reader

normanb, 2024-10-23:
I am using the Capella Open Data in particular this SLC tiff

https://capella-open-data.s3.amazonaws.com/data/2024/10/11/CAPELLA_C15_SP_SLC_HH_20241011085812_20241011085836/CAPELLA_C15_SP_SLC_HH_20241011085812_20241011085836.tif

I convert this tiff file to uncompressed with `gdal_translate co TILED=no CAPELLA_C15_SP_SLC_HH_20241011085812_20241011085836.tif capella_uncompressed.tif`

And open this with sarpy

```
from sarpy.io.complex.converter import open_complex
rdr = open_complex("capella_uncompressed_slc.tif")
```

Which gives this error

```
  File "xxx/python3.10/site-packages/sarpy-1.3.58-py3.10.egg/sarpy/io/complex/converter.py", line 106, in open_complex
    reader = opener(file_name)
  File "xxx/python3.10/site-packages/sarpy-1.3.58-py3.10.egg/sarpy/io/complex/capella.py", line 516, in is_a
    return CapellaReader(capella_details)
  File "xxx/python3.10/site-packages/sarpy-1.3.58-py3.10.egg/sarpy/io/complex/capella.py", line 470, in __init__
    sicd = self.capella_details.get_sicd()
  File "xxx/python3.10/site-packages/sarpy-1.3.58-py3.10.egg/sarpy/io/complex/capella.py", line 411, in get_sicd
    ss_zd_s = img['image_geometry']['delta_line_time']
KeyError: 'delta_line_time'
```

Is there a specification for the Capella SLC product so I can check whether `delta_line_time` is a required parameter and put a PR for the fix together? Or is this a data product issue?

bombaci-vsc, 2024-10-23:
The specification can be found on Capella's site: https://support.capellaspace.com/hc/en-us/articles/5607458273940-SAR-Imagery-Products-Format-Specification

Capella provides SICD which can be read by SARPy's SICD reader.  For example, the SICD for this dataset can be found at https://capella-open-data.s3.amazonaws.com/data/2024/10/11/CAPELLA_C15_SP_SICD_HH_20241011085812_20241011085836/CAPELLA_C15_SP_SICD_HH_20241011085812_20241011085836.ntf

As SICD is directly available, the Capella Tiff converter has not been actively maintained.  This dataset is processed as PFA, which is not supported by the Capella converter.

normanb, 2024-10-23:
The SICD file is parsed fine, thank you for the clarification and the link to the product spec.

I understand that Capella (and Umbra) use PFA for image formation, as SarPy retrieves the image data when you say PFA is not supported by the Capella Converter what are you referring to? Are you referring to the retrieval of [PFA metadata](https://sarpy.readthedocs.io/en/latest/io/complex/sicd_elements/pfa.html) with the Capella TIFF converter?

bombaci-vsc, 2024-10-23:
The modules under `sarpy.io.complex` parse complex images and convert native metadata to SICD metadata.  In many cases this conversion is not trivial and involves computation, not just retrieval.  `capella.py` was written before Capella produced SLCs with `type==pfa` and as such it does not have logic for converting into SICD's PFA metadata.

petersontex, 2025-05-07:
Closing this issue, given the following:

"Capella provides SICD which can be read by SARPy's SICD reader."

closed 2025-05-07 by petersontex