ngageoint/sarpy #576: CPHDReader from remote file

holmesv3, 2025-01-07:
The documentation has an example for SICD and SIDD readers being created from a remote file using `smart_open`. When trying the same with a `CPHDReader` it fails because a local file path is expected. 

Are there plans to add remote file capability for the CPHDReader, or is there an implementation detail which blocks it?

dhaverporth, 2025-01-13:
Hello,  we are currently working on SARPy 2.0 which  I believe will address this issue.  We are currently calling the new release SARKit.  You can check out the new release here:  [(https://github.com/ValkyrieSystems/sarkit)]  Please keep in mind that this code is not finished and is still experimental.

holmesv3, 2025-01-13:
Thanks for the response! Is sarpy going to continue being maintained, or is this new sarkit replacing it? Should new features/bugfixes be directed towards sarkit instead?

petersontex, 2025-06-18:
@holmesv3 We will continue to maintain sarpy for the time being. Please submit new feature and bugfix requests here. We will review them and determine if the request should be resolved in sarpy or in the future capability (sarkit)

I am forwarding this capability request to sarkit. 

sarkit request here: https://github.com/ValkyrieSystems/sarkit/issues/45