ngageoint/sarpy #578: SIDD 2.0 Compression Schema Reading Bug

ktopolov, 2025-01-18:
It seems there might be a bug in XML metadata reading for SIDD2.0+ for the "Compression" field. It seems from the specification that this is the expected formatting:
```
    <Compression>
        <J2K>
            <Original>
                <NumWaveletLevels>6</NumWaveletLevels>
                <NumBands>1</NumBands>
                <LayerInfo numLayers="2">
                    <Layer index="1">
                        <Bitrate>2.00000000000000000E01</Bitrate>
                    </Layer>
                    <Layer index="2">
                        <Bitrate>2.00000000000000000E01</Bitrate>
                    </Layer>
                </LayerInfo>
            </Original>
        </J2K>
    </Compression>
```
and C++ Six library writes it this way, but sarpy fails to read with the error:
```
Attribute Custom of class FilterKernelType received <Element '{urn:SIDD:2.0.0}Custom' at 0x7f101f671b20>,
	but values ARE REQUIRED to be one of ('GENERAL', 'FILTER BANK')
Failed converting <Element '{urn:SIDD:2.0.0}Original' at 0x7f101f6997b0> of type <class 'xml.etree.ElementTree.Element'> to Serializable type <class 'sarpy.io.product.sidd2_elements.Compression.J2KSubtype'>
	for field Original of class J2KType with exception <class 'KeyError'> - 'size'.
	Setting value to None, which may be against the standard.
```
It seems that `sarpy` is instead expecting `numLayers` to instead be `size`, or it may use the wrong "descriptor" type altogether - it appears to simply want an array of bitrates rather than these individual indexed layers.

ARKAMarkDologuin, 2025-03-03:
I've added a parser that goes through to the different Layers and gets the bitrates to place into an array as per LayerInfo object expectations
https://github.com/ARKAMarkDologuin/sarpy/commit/e5ab5e39963f659acccb5c51d21e407dbd509888

petersontex, 2025-12-02:
Please test this with the new v1.3.62 release candidate. 

The code is here: https://github.com/ngageoint/sarpy/releases/tag/v1.3.62rc1

Also available from PyPi (As a full release) here:  https://pypi.org/project/sarpy/

closed 2025-12-09 by petersontex