ngageoint/sarpy #581: CRSD Receive Only Type

RyanUmby, 2025-02-20:
Writing a passive CRSD covariance information raises an error, the offending code [here](https://github.com/ngageoint/sarpy/blob/master/sarpy/io/received/crsd1_elements/ErrorParameters.py#L319).

I believe the `_fields` and `_required` should be `ReceiveSensor` and not `ReceiveSensorType`. Furthermore, the type associated with the attribute should be `ReceiveSensorType` and not `RadarSensorType`

pressler-vsc, 2025-02-20:
It may not be relevant if you're dealing with DRAFT2021 CRSDs, but I wanted to note that CRSD has been significantly reworked (e.g. arbitrary transmit waveform descriptions, transmit-only files) in the intervening time.

The [schema for the most recent draft](https://github.com/ValkyrieSystems/sarkit/blob/main/sarkit/crsd/schemas/NGA.STND.0080-1_1.0_DRAFT_CRSD_DIDD_2024_12_30.xsd) (2024-12-30) as well as code for reading/writing/consistency checking is available in a new project: [SARkit](https://github.com/ValkyrieSystems/sarkit). SARkit is in early development, so the API is rapidly evolving as features are being added,but the 2024-12-30 draft will likely be much closer to the CRSD v1.0 anticipated for NSG adoption in 2025.

petersontex, 2025-06-18:
User was directed to sarkit by @pressler-vsc. Issue is resolved.

closed 2025-06-18 by petersontex