ngageoint/sarpy #594: Non backwards compatible change to create_sidd_structure

bombaci-vsc, 2025-08-04:
v1.3.61 breaks compatibility with v1.3.60 due to the change of `create_sidd_structure_v[2|3]` in https://github.com/ngageoint/sarpy/commit/61da4cb76db413c9b15e201ac1a1823dc86a3d97 .

The `remap_function` argument was added with a default of `None`.  However, the body of the function assumes it isn't None.  see:

https://github.com/ngageoint/sarpy/blob/master/sarpy/processing/sidd/sidd_structure_creation.py#L247

petersontex, 2025-08-15:
@bombaci-vsc We believe we have resolved this issue. 

Please test out release v1.3.61.1 to confirm that this resolves your problem.

If this patch does not resolve your problem please respond with a detailed example of how you are trying to use the code.

The release is available here: 

https://github.com/ngageoint/sarpy/releases/tag/v1.3.61.1

or from PyPi here:

https://pypi.org/project/sarpy/

petersontex, 2025-08-22:
No response, closing issue.